BrewBlox/brewblox-firmware #115: In temperature profile, support starting a temperature step once a temperature has been reached

elcojacobs, 2019-07-07:


steersbob, 2022-05-11:
Updated the issue with the following changes:

- Changed WAIT_ABSOLUTE to WAIT_UNTIL.
- Added WAIT_TEMPERATURE, WAIT_TEMPERATURE_ABOVE, WAIT_TEMPERATURE_BELOW instructions that directly target a temperature sensor.
- Changed SET_ANALOG / WAIT_ANALOG to SET_PWM / WAIT_PWM to be more explicit.
- Instruction arguments are now formatted as `key=value`. This makes the syntax more explicit, and allows for deprecating or adding arguments.
- Removed settling time for the WAIT_SETPOINT instruction.
- Moved error status enum values to the `error` field.
- Added TARGET_DISABLED error value.
- Changed UTC_NOT_AVAILABLE to SYSTEM_TIME_NOT_AVAILABLE.
- Added `disabledAt` and `disabledDuration` fields.
- Explicitly mentioned that the syntax uses abbreviated temperature units.
- Explicitly mentioned that single quotes are optional for all arguments that do not contain spaces.

elcojacobs, 2022-05-19:
Persisting active instruction to eeprom can be a problem with fast repeating sequences.
Those should only persist to backup ram, but how do we distinguish?

steersbob, 2022-05-25:
- Converted reset field to `overrideState` + state fields.
- Removed WAIT_PWM
- Added WAIT_TEMP_NOT_BETWEEN
- Added WAIT_TEMP_UNEXPECTED
- Shortened WAIT_TEMPERATURE_XXXXX instructions to WAIT_TEMP_XXXXX

PR #387 (2022-05-25): Sequence block
Resolves #115
closed 2022-05-25 by steersbob